ipa320/cob_robots #259: Question, not issue: about gripper control in simulation

TheDash, 2015-01-15:
Sorry if this isn't the right forum for this. I'm curious as to how the architecture of your gripper control looks like? Is it multi-dof? What plugins do you use for gazebo?

I'm trying to make the 3 finger roboteq s-model claw work in gazebo, but need a controller for it. Did you guys manage to make a multi-dof thing work with the jointtrajectory effort controller?

ipa-fmw, 2015-01-16:
I assume you're refering to the Care-O-bot 3 gripper which is a Schunk SDH. For gazebo we modeled each joint as a revolute joint and added a joint trajectory controller to control all 7 joints in one action.
see 
URDF: https://github.com/ipa320/schunk_modular_robotics/tree/indigo_dev/schunk_description/urdf/sdh
controller: https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_controller_configuration_gazebo/controller/gripper/gripper_controller_sdh.yaml
launch file: 
```
	<!-- Gripper -->
	<include file="$(find cob_controller_configuration_gazebo)/launch/components/generic_controller.launch">
		<arg name="component_name" value="gripper"/>
		<arg name="version" value="sdh"/>
	</include>
```

ipa-fxm, 2015-01-16:
There are definitely better forums for such kind of questions, e.g. [answers.gazebo](http://answers.gazebosim.org/questions), [answers.ros](http://answers.ros.org/questions/) or the [ros_control SIG](http://wiki.ros.org/sig/RobotControl)

Just two more hints:
Why do you want to use effort controller? Is the actual hardware controlled by sending efforts, too?
If you want to just control the joint positions of the gripper joints, try using a JointPositionHardwareInterface and the JointPositionControllers in gazebo. By doing this, you don't need to find PID values as you can use a forward command controller.

Could you please close this issue and ask further questions on the forums mentioned above?

closed 2015-01-16 by ipa-fmw