ipa320/cob_robots #268: [Indigo] Restructure and cleanup

ipa-fxm, 2015-02-10:
It's currently not possible to find a clean structure for the launch and parameter files that set up all robots (cob3-6, cob3-9, cob4-1, cob4-2, cob4-6) in both real hardware and simulation with the new namespace layout under indigo!

We will need to review this 
- after the new base controller plugins are finalized
- the cob3-X robots hardware is set up for indigo

@ipa-fmw @ipa-nhg @ipa-mig @ipa-mdl

ipa-fxm, 2015-02-12:
I started to use the new namespace structure for cob3-6 in simulation
[cob_common/cob3-updates](https://github.com/ipa-fxm/cob_common/tree/cob3-updates)
[schunk_modular_robotics/cob3-updates](https://github.com/ipa-fxm/schunk_modular_robotics/tree/cob3-updates)
[cob_robots/cob3-6_updates](https://github.com/ipa-fxm/cob_robots/tree/cob3-6_updates)

It's not yet finished.....this is just to let you know about it.....will be finalized and synchronized with indigo hardware for cob3's before a PullRequest to ipa320 is sent...

ipa-fmw, 2015-03-16:
still need to fix launch files for schunk_sdh, cob_trajectory_controller and head_axis namespaces in cob_bringup launch files

see https://github.com/ipa320/schunk_modular_robotics/issues/131, https://github.com/ipa320/schunk_modular_robotics/issues/130, https://github.com/ipa320/cob_driver/issues/208, https://github.com/ipa320/cob_control/issues/26

ipa-fxm, 2015-03-18:
more restructuring done in #278 #279 

adaptions to schunk_powercube_chain, schunk_sdh, cob_trajectory_controller and cob_head_axis done in:
 - https://github.com/ipa320/schunk_modular_robotics/pull/132
 - https://github.com/ipa320/cob_driver/pull/210
 - https://github.com/ipa320/cob_control/pull/27

ipa-fxm, 2015-04-20:
I guess most of the restructuring is done and merged...
It performs well on cob4 and simulation....
However, there still seems to be some mismatches on cob3's (see https://github.com/ipa320/cob_command_tools/issues/93)

ipa-fxm, 2015-06-14:
@ipa-nhg @ipa-fmw 
now that https://github.com/ipa320/cob_robots/pull/303 has been merged should we adjust cob3-6 and cob3-9 for indigo accordingly, i.e. bringup launch file and configs?

ipa-fxm, 2015-09-04:
#314 probably fixes it in a satisfactory way for now....

closed 2015-09-04 by ipa-fxm