ipa320/cob_robots #283: add mimic launch file in cob_bringup

ipa-fmw, 2015-03-19:
see https://github.com/ipa320/cob4/issues/179

ipa-nhg, 2015-03-20:
https://github.com/ipa320/cob_robots/pull/285

ipa-fxm, 2015-04-20:
what's the status here?
I've seen the launch file on the robot but it's not in the repository yet....probably due to the [comments](https://github.com/ipa320/cob_robots/pull/285#discussion-diff-26847895)?

ipa-fxm, 2015-05-01:
As the [comment](https://github.com/ipa320/cob_robots/pull/285#discussion-diff-26847895) about unifying topic/service/action names of mimic, sound andd lights is considered in https://github.com/ipa320/cob_driver/pull/223, https://github.com/ipa320/cob_robots/pull/296 and https://github.com/ipa320/cob_command_tools/pull/98, this can be closed

closed 2015-05-01 by ipa-fxm