ipa320/cob_robots #295: robot_state_publisher

jobstvogt, 2015-04-22:
I have some tf problems (TF_OLD_DATA, etc.) and I think it is related to the change that every component has its own robot state publisher. Why do you think it is better to have so many (in my case 7) robot state publishers that are all publishing the same transforms?

When I run the tf_monitor i get this strange output with unknown publisher:

RESULTS: for all Frames

Frames:
Frame: /base_footprint published by unknown_publisher Average Delay: 0.185526 Max Delay: 0.18663
Frame: arm_0_link published by unknown_publisher Average Delay: -0.312836 Max Delay: 0
Frame: arm_1_link published by unknown_publisher Average Delay: 0.188703 Max Delay: 0.194755
Frame: arm_2_link published by unknown_publisher Average Delay: 0.188705 Max Delay: 0.194758
Frame: arm_3_link published by unknown_publisher Average Delay: 0.188706 Max Delay: 0.194759
Frame: arm_4_link published by unknown_publisher Average Delay: 0.188706 Max Delay: 0.19476
Frame: arm_5_link published by unknown_publisher Average Delay: 0.188707 Max Delay: 0.194761
Frame: arm_6_link published by unknown_publisher Average Delay: 0.188707 Max Delay: 0.194762
Frame: arm_7_link published by unknown_publisher Average Delay: 0.188708 Max Delay: 0.194763
Frame: base_laser_front_link published by unknown_publisher Average Delay: -0.312834 Max Delay: 0
Frame: base_laser_rear_link published by unknown_publisher Average Delay: -0.312833 Max Delay: 0
Frame: base_laser_top_link published by unknown_publisher Average Delay: -0.312832 Max Delay: 0
Frame: base_link published by unknown_publisher Average Delay: -0.312834 Max Delay: 0
Frame: bl_caster_r_wheel_link published by unknown_publisher Average Delay: 0.18574 Max Delay: 0.187387
Frame: bl_caster_rotation_link published by unknown_publisher Average Delay: 0.185742 Max Delay: 0.187388
Frame: br_caster_r_wheel_link published by unknown_publisher Average Delay: 0.185743 Max Delay: 0.187389
Frame: br_caster_rotation_link published by unknown_publisher Average Delay: 0.185743 Max Delay: 0.187389
Frame: cam3d_depth_frame published by unknown_publisher Average Delay: -0.312831 Max Delay: 0
Frame: cam3d_depth_optical_frame published by unknown_publisher Average Delay: -0.31283 Max Delay: 0
Frame: cam3d_link published by unknown_publisher Average Delay: -0.31283 Max Delay: 0
Frame: cam3d_rgb_frame published by unknown_publisher Average Delay: -0.312829 Max Delay: 0
Frame: cam3d_rgb_optical_frame published by unknown_publisher Average Delay: -0.312828 Max Delay: 0
Frame: fl_caster_r_wheel_link published by unknown_publisher Average Delay: 0.185744 Max Delay: 0.18739
Frame: fl_caster_rotation_link published by unknown_publisher Average Delay: 0.185745 Max Delay: 0.187391
Frame: fr_caster_r_wheel_link published by unknown_publisher Average Delay: 0.185746 Max Delay: 0.187391
Frame: fr_caster_rotation_link published by unknown_publisher Average Delay: 0.185746 Max Delay: 0.187392
Frame: head_axis_link published by unknown_publisher Average Delay: 0.185114 Max Delay: 0.185585
Frame: head_base_link published by unknown_publisher Average Delay: -0.312828 Max Delay: 0
Frame: head_cam_reference_link published by unknown_publisher Average Delay: -0.312827 Max Delay: 0
Frame: head_cover_link published by unknown_publisher Average Delay: -0.312826 Max Delay: 0
Frame: lookat_base_link published by unknown_publisher Average Delay: -0.312825 Max Delay: 0
Frame: pg70_finger_left_link published by unknown_publisher Average Delay: 0.18832 Max Delay: 0.188673
Frame: pg70_finger_right_link published by unknown_publisher Average Delay: 0.188322 Max Delay: 0.188675
Frame: pg70_gripper_link published by unknown_publisher Average Delay: -0.312824 Max Delay: 0
Frame: torso_base_link published by unknown_publisher Average Delay: -0.312824 Max Delay: 0
Frame: torso_lower_neck_tilt_link published by unknown_publisher Average Delay: 0.188553 Max Delay: 0.189466
Frame: torso_pan_link published by unknown_publisher Average Delay: 0.188555 Max Delay: 0.189468
Frame: torso_upper_neck_tilt_link published by unknown_publisher Average Delay: 0.188556 Max Delay: 0.18947
Frame: tray_base_link published by unknown_publisher Average Delay: -0.312823 Max Delay: 0
Frame: tray_left_link published by unknown_publisher Average Delay: -0.312822 Max Delay: 0
Frame: tray_link published by unknown_publisher Average Delay: 0.188313 Max Delay: 0.188757
Frame: tray_phidgets_1_link published by unknown_publisher Average Delay: -0.312821 Max Delay: 0
Frame: tray_phidgets_2_link published by unknown_publisher Average Delay: -0.312821 Max Delay: 0
Frame: tray_phidgets_3_link published by unknown_publisher Average Delay: -0.31282 Max Delay: 0
Frame: tray_phidgets_4_link published by unknown_publisher Average Delay: -0.312819 Max Delay: 0
Frame: tray_right_link published by unknown_publisher Average Delay: -0.312819 Max Delay: 0

All Broadcasters:
Node: unknown_publisher 571.44 Hz, Average Delay: -0.0757929 Max Delay: 0.19476

And roswtf:

ERROR TF multiple authority contention:
 * node [/tray/robot_state_publisher] publishing transform [arm_0_link] with parent [base_link] already published by node [/torso/robot_state_publisher]
 * node [/tray/robot_state_publisher] publishing transform [base_link] with parent [base_footprint] already published by node [/torso/robot_state_publisher]
 * node [/tray/robot_state_publisher] publishing transform [base_laser_front_link] with parent [base_link] already published by node [/torso/robot_state_publisher]
 * node [/tray/robot_state_publisher] publishing transform [base_laser_rear_link] with parent [base_link] already published by node [/torso/robot_state_publisher]
 * node [/tray/robot_state_publisher] publishing transform [base_laser_top_link] with parent [base_link] already published by node [/torso/robot_state_publisher]
 * node [/tray/robot_state_publisher] publishing transform [cam3d_depth_frame] with parent [cam3d_link] already published by node [/torso/robot_state_publisher]
 * node [/tray/robot_state_publisher] publishing transform [cam3d_depth_optical_frame] with parent [cam3d_depth_frame] already published by node [/torso/robot_state_publisher]

and so on...

ipa-fmw, 2015-04-23:
we identified that too. The reason is that the tf tree for Care-O-bot is quite big and with multiple robot_state_publishers distributed over several pcs this will cause a lot of network traffic.

We've worked out a solution using the joint_state_publisher package and use that to aggregate all joint_states topics into a global /joint_states and then only use one robot_state_publisher again. This will be merged to ipa320/cob_robots soon.

ipa-fxm, 2015-04-29:
see https://github.com/ipa-fmw/cob_robots/commit/6dfb10885bfffde8f72c487827948a3dc1265a2a
and https://github.com/ipa-fmw/cob_robots/commit/8ccfe5f351756431d28130ef27727e07d5b0ab86

ipa-fmw, 2015-04-30:
included in https://github.com/ipa320/cob_robots/pull/296

closed 2015-05-01 by ipa-fxm