ipa320/cob_robots #309: Replace controller_manager spawner

ipa-mdl, 2015-06-18:
The controller_manager spawner script does not really suit our use case with deferred initialization.
As mentioned in https://github.com/ros-controls/ros_control/pull/188 we should just use the controller_manager script.
The spawner is meant for on-demand start of controllers via launch files with life cycle management.

So we can achieve the desired behaviour with:
```
<node ns="$(arg component)" name="$(arg component)_controller_spawner" pkg="controller_manager" type="controller_manager" args="spawn joint_state_controller joint_trajectory_controller" respawn="false" output="screen" />
```
This node just waits (indefinitely) for the controller manager services, calls load and start for the list of controllers, and exits.

ipa-fxm, 2015-06-20:
much better!
- https://github.com/ipa320/schunk_robots/pull/49
- https://github.com/ipa320/cob_test_robots/pull/5
- https://github.com/ipa320/cob_robots/pull/310

ipa-fxm, 2015-06-29:
done

closed 2015-06-29 by ipa-fxm