ipa320/cob_robots #320: raw3-3 roslaunch check failure

ipa-fmw, 2015-08-03:
#300 introduced an include in raw3-3.xml which opens a dependency to rplidar_ros.

Although this is not used by default, the roslaunch check for cob_bringup fails now http://jenkins.ros.org/job/devel-indigo-cob_robots/ARCH_PARAM=amd64,UBUNTU_PARAM=trusty,label=devel/57/testReport/(root)/Results/test_ran/.
To avoid that we should add an empty package rplidar_ros to cob_substitute. @ipa-frm can you please add this and send a PR?

ipa-fmw, 2015-08-03:
@ipa-frm you don't need to get active here, sorry.
@ipa-fxm will resolve this in https://github.com/ipa320/cob_substitute/pull/29/

ipa-fxm, 2015-08-03:
fixed via https://github.com/ipa-fxm/cob_substitute/commit/4da4497dd12991b13d3983339aa2576fd7d56210 in https://github.com/ipa320/cob_substitute/pull/29

However, `cob_substitute` so far substituted only our own private repositories, with `rplidar_ros` this an official ROS package (not under ipa320 supervision) that just hasn't been released yet...
There might be a conflict in case we release `cob_substitute` and `rpllidar_ros` is going to be released eventually into Indigo

ipa-fmw, 2015-08-04:
other question (to @ipa-frm and @ipa-mig): what is preventing us (or the maintainer) to release `rplidar_ros` under indigo? Did somebody try to trigger the maintainer to release it?

ipa-mig, 2015-08-04:
@ipa-fmw There has been a release for hydro.
So maybe we can try to contact them and ask for a new release. I'll do that.

On the other hand, they seem to be pretty inactive. The last commit has been on 21st of April, there are two PRs open (from which at least one makes sense) for over a month and no feedback from them.
The question is, should we try to take this over? I.e. additional effort on our side, especially if they start to release new features?

ipa-mig, 2015-08-04:
@ipa-fmw @ipa-fxm 
Set up an issue at robopeak/rplidar_ros#15

ipa-fmw, 2015-08-04:
Let's try to trigger them first. If there is no Motivation for an indigo release or no activity, we can at least offer our support with releasing the current state. This does not require to take over full maintainership

ipa-mig, 2015-08-04:
Done already, see link above.

ipa-fxm, 2015-09-04:
What's the status here?
Close? Help @robopeak with the release?

ipa-mig, 2015-09-07:
Well, an `rplidar_ros` release would be nice. However, @robopeak is not responding to any questions and they haven't been active sind April.

Should we just go ahead with a release?

This might actually now become a problem with the latest `cob_substitute` release...

ipa-fxm, 2015-11-20:
I guess we can all live with the current situation. Thus, I'll close this issue!

closed 2015-11-20 by ipa-fxm