ipa320/cob_robots #324: cob_simulation install from dep issue

ipa-frm, 2015-08-14:
after apt-get installing cob_simulation and starting with ```roslauch cob_bringup_sim robot.launch``` the following error occurs: 
```ERROR: cannot launch node of type [cob_controller_configuration_gazebo/gazebo_services.py]: can't locate node [gazebo_services.py] in package [cob_controller_configuration_gazebo] ```

Tried to start the node with ```rosrun cob_controller_configuration_gazebo gazebo_services.py ```, which leads to the following error:
```[rosrun] Couldn't find executable named gazebo_services.py below opt/ros/indigo/share/cob_controller_configuration_gazebo ```

ros seems to look for the executable in ```/opt/ros/indigo/share/cob_controller_configuration_gazebo``` but it is actually located under ```/opt/ros/indigo/lib/cob_controller_configuration_gazebo/src```

ipa-nhg, 2015-08-14:
are you sourcing /opt/ros/indigo/setup.bash? have you or had you got overlays on your sourced workspace? did you remove the folders build , devel and install after the installation and before a new catkin make? Also a 'rosdep update' can help

ipa-fxm, 2015-08-15:
I asked @ipa-frm to report this issue.
Now I tested it on my laptop....and there, I can't reproduce this error...

What I did is:
[image]

Summary:
I only `source /opt/ros/indigo/setup.bash`.
`roscd cob_controller_configuration_gazebo` also brings me to `/opt/ros/indigo/share/cob_controller_configuration_gazebo`
The scripts are located in `/opt/ros/indigo/lib/cob_controller_configuration_gazebo/src` (same as @ipa-frm)
But `rosrun cob_controller_configuration_gazebo` works for me!

@ipa-frm can you please re-check your installed version?
`dpkg-query -s ros-indigo-cob-controller-configuration-gazebo` should print the following
```
Package: ros-indigo-cob-controller-configuration-gazebo
Status: install ok installed
Priority: extra
Section: misc
Installed-Size: 154
Maintainer: Florian Weisshardt <fmw@ipa.fhg.de>
Architecture: amd64
Version: 0.6.3-0trusty-20150730-0951-+0000
Depends: ros-indigo-cob-bringup, ros-indigo-cob-cam3d-throttle, ros-indigo-cob-phidgets, ros-indigo-cob-script-server, ros-indigo-diagnostic-msgs, ros-indigo-rospy, ros-indigo-rostopic, ros-indigo-std-msgs, ros-indigo-std-srvs, ros-indigo-topic-tools
Description: This packages provides the controller configuration for simulating Care-O-bot in the gazebo simulator.
Homepage: http://ros.org/wiki/cob_controller_configuration_gazebo
```

ipa-fxm, 2015-08-15:
However, [this](http://docs.ros.org/api/catkin/html/howto/format2/installing_python.html) install tag for python scripts is different from what we use

```
catkin_install_python(PROGRAMS scripts/your_node1 scripts/your_node2
                      DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION})
```

vs. 

```
install(PROGRAMS src/fake_diagnostics.py src/gazebo_services.py src/gazebo_topics.py
  DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}/src
)
```

ipa-frm, 2015-08-17:
seems to be aproblem with my workspace, so I'll close this. Thanks @ipa-fxm for looking after it :smiley:

closed 2015-08-17 by ipa-frm