ipa320/cob_robots #348: review camera topics

ipa-fmw, 2015-11-04:
taken from #347: 
cam3d topic names need a significant review anyway as topics in simulation do not match the topic names on real hardware for most of the kinect/softkinetic/prosilica sensors....

As #346 tries to introduce a consistent naming in simulation...can we tackle this issue after #346 has been merged and then try to remove trailing slashes for link_names wherever it applies?

ipa-fxm, 2015-11-21:
The `cob_common` part has been started in https://github.com/ipa320/cob_common/pull/179
...but it is not yet finalized....`cob_robots` is still missing...and wherever this change might have consequences (i.e. perception, moveit, rviz configs, ...)

ipa-fxm, 2016-02-05:
here is a comparison of the current topic from a cam3d in simulation and hw:
https://gist.github.com/ipa-fxm/d37e2e3f7e7ce9f8884f

ipa-fxm, 2016-02-05:
I already looks quite similar...still there are a few things, I would (personally) would like to have clarified by some perception experts/users @ipa-rmb @ipa-josh 

In simulation, we have `depth/image_raw`, but `depth_registered/camera_info` and `depth_registered/points`. Shouldn't the `image_raw` topic always be in the same namespace as the respective `camera_info` topic?
Would you want to have all three topic within `depth` namespace or all three topics within `depth_registered` namespace or would you want all six topic?
Is it possible to compose a list of topics from the hw list that we actually do subscribe somewhere? I.e. if you are to choose topics from that list, which would you a) definitely need and b) definitely not need

ipa-rmb, 2016-02-08:
Here are some answers to your questions from josh and me:
1. depth/image_raw is not used by josh or me. It would probably make more sense to rename it to the depth_registered namespace if the real hardware driver does the same.
2. We need all six variants, the three in depth name space and the other three from depth_registered.
3. Has not been used by us so far.
4. Usually the topics rgb/image (or rgb/image_raw or rgb/image_color -> depends on driver version) and rgb/camera_info and depth/camera_info, depth/points and depth_registered/camera_info, depth_registered/points are needed. All compressed topics are probably never used but they are created automatically anyways whenever a image_transport is used for image transmission. However, I see limited use of this list because the names change frequently in the original driver, so we are used to adapt topics anyways.

I cannot provide a list for stereo cameras since I have not used them so far. Obviously I would expect that the camera_info and image (or image_raw or image_color) topics would be most important.

ipa-fxm, 2016-02-09:
I think, the gazebo_plugin for the camera can only provide __one__ namespace for depth images and __one__ for colored images....
Thus, I would suggest to go for `depth_registered` for the depth images and `rgb` for the colored images!

ipa-fxm, 2016-02-09:
I just provided a PR in https://github.com/ipa320/cob_common/pull/183

closed 2016-02-10 by ipa-fmw