ipa320/cob_robots #366: [cob_default_robot_behavior] fix service response

ipa-fmw, 2015-11-16:
currently returns

```
[INFO] [WallTime: 1447667136.528934] <<torso_front_home>> <<behavior>>
Unhandled exception in thread started by <rospy.impl.tcpros_service.ServiceProxy object at 0x7f1a7df0b3d0>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 696, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/behavior/torso_front_home] responded with an error: service cannot process request: service handler returned None
```

ipa-fmw, 2015-12-11:
solved in #393

ipa-fxm, 2016-02-04:
solved in #406

closed 2016-02-04 by ipa-fxm