ipa320/cob_robots #374: Add launch file checks and URDF checks

ipa-fxm, 2015-11-20:
We need to write these tests and add them to travis

@ipa-fmw @ipa-mdl 
Do you have a good idea how to deal with the problem about using `$(find )` and `$(arg )` and such mixed together?

ipa-fxm, 2015-11-20:
@ipa-mdl:
rosinstall template https://github.com/ipa320/setup/blob/master/repositories/cob_indigo_bringup.rosinstall

ipa-fxm, 2015-11-21:
Sweet!
Let's bring this new travis concept to the other repos as well - after the merge session on Tuesday!

ipa-fxm, 2015-11-21:
We might need to update/recheck the `robotlist` entries in e.g. `cob_bringup`, `cob_controller_configuration_gazebo`, `cob_gazebo`,...

ipa-mdl, 2015-11-21:
It would be even better to keep a consolidated list at one single place. It could be a symlinked cmake files that is included or we could keep the list in a  dedicated package (e.g. cob_robot_list) and expose it to dependent packages (`CFG_EXTRAS`).

PR #375 (2015-11-23): modified travis config to build with source overlays
closes #374
closed 2015-11-23 by ipa-fmw