ipa320/cob_robots #403: raw3-1 emergency state topic in wrong ns

ipa-srd, 2016-01-19:
in raw3-1.xml the relayboard.launch is started in simulation mode which results in a wrong ns of the /emergency_stop_state topic (which is relayboard/emergency_stop_state) and the robot can not be moved.

ipa-nhg, 2016-01-19:
This PR https://github.com/ipa320/cob_robots/pull/395  solves the problem

ipa-fxm, 2016-01-19:
Even if #395 solves the namespace problem in simulation....why is this node run in simulation __on real hardware__ anyways?

ipa-mig, 2016-01-19:
I guess because the r@w does not have a relayboard. How is it handled on cob4?

ipa-fxm, 2016-01-19:
If raw3-1 does not have a relayboard, we should not have the robot to require a topic coming from this node in order to be able to operate it...

destogl, 2016-01-20:
> Even if #395 solves the namespace problem in simulation....why is this node run in simulation mode on real hardware anyways? There seems to be something wrong about this...

This is something I found out a year a go or so, but I was never really sure and had time to change it for all robots. For raw3-5 should be changed in #400 

> I guess because the r@w does not have a relayboard.

raw3-5 has? Or at least can publish this topics through phidgetboard. Probably I have to look at it again more closely.... Anyway I solved it for raw3-5...

ipa-mig, 2016-01-20:
Yes, the `emergency_stop_state`  topic can be published from a phidget board.

destogl, 2016-01-20:
But than on all r@ws (if all have phidged board) this:

```
<include file="$(find cob_bringup)/drivers/relayboard.launch" >
		<arg name="sim" value="true" />
</include>
```

should be changed for this:
```
<include file="$(find cob_bringup)/drivers/phidgets.launch" >
		<arg name="robot" value="$(arg robot)" />
</include>
```

as I did in #400

ipa-mig, 2016-01-21:
True, but not all have a phidget board (actually, I'm not sure if there not only two that actually have one). And there are different versions. We have to check with our r@w's, at least the two that are still at IPA. For the rest, we need to keep it as is.

destogl, 2016-01-21:
Ok check it, at KIT we have phidgetboard, so for raw3-5 it can be also changed.

ipa-bnm, 2016-01-21:
from my point of view just two things need to be done.
1. merge PR #395 because we still need the simulated relayboard for robots without relayboard and phidgetboard.
2. replace relayboard launch with phidgetboard launch for raw3-1 and raw3-5. raw3-3 was already changed and the other raw's do not have a phidget board.

ipa-fxm, 2016-01-21:
> because we still need the simulated relayboard for robots without relayboard and phidgetboard.
Why? Which node requires the topic to be available?
Is it only me who thinks it is strange that a (software) node (I don't know which one it is) requires a hardware to be present in order to work...?
I thought we use "simulated" nodes in order to have the same set of topics and message in __simulation__ than we do on the respective hardware. __NOT__ for making all hardware setups equal in case one does not have a relayboard...

ipa-mig, 2016-01-22:
Alright. The `/emergency_stop_state` topic is required by the `cob_undercarriage_control` node ([subscriber callback](https://github.com/ipa320/cob_driver/blob/indigo_dev/cob_undercarriage_ctrl/ros/src/cob_undercarriage_ctrl.cpp#L315-L344) / [setter function in controller](https://github.com/ipa320/cob_driver/blob/indigo_dev/cob_undercarriage_ctrl/common/src/UndercarriageCtrlGeom.cpp#L709-L731) / [block calculation of control step if in EM stop](https://github.com/ipa320/cob_driver/blob/indigo_dev/cob_undercarriage_ctrl/common/src/UndercarriageCtrlGeom.cpp#L368-L372)).

I guess this is a left over from the `cob3`-era, where all `cob3`'s had a relayboard.
I'm not sure if this is used anywhere else (diagonstics? @ipa-fmw, `cob_omni_drive_controller`? @ipa-mdl) or how the controllers behave, if we where to remove this.

If the `emergency_stop_state` is set to `true`, the output commands will be 0.0 and some internal controller state variables will be reset. If this, however, is required, and how the control behaviour will change then, I cannot say. Maybe worth a test.

ipa-bnm, 2016-01-22:
as @ipa-mig wrote, `cob_undercarriage_control` needs the `emergency_stop_state` topic to be published. If it's not published the base will not move. This behavior is implemented because of safety reasons. (No EM Stop State means that there is probably something wrong and we should not move the base).

ipa-fxm, 2016-01-22:
Yes, the base should not be moved in case of EM-Stop, but this is enforced due to the EM-Stop cutting of the power from the drives - on the hardware-side!
It should not need to be handled by software...even more, the hardware will not start to move after the EM-Stop has been released again without recovering it anyways...

ipa-mig, 2016-01-22:
Yes to both. It would be nice to have this topic and handle the different controllers respectively.
Especially, as it allows for differntiating between button stop and scanner stop.

However, if the sim node is used, the emergency stop flags are always `false`, i.e. not in EM stop. Thus, I guess it could be removed.

ipa-mdl, 2016-01-22:
> It should not need to be handled by software...even more, the hardware will not start to move after the EM-Stop has been released again without recovering it anyways...

The software controller should reset its internals in order to not jump after recover.
Within the migration of the platform controllers to ROS (for cob3) the `emergency_stop_state` was introduces and the EM stop was handled in (too) many places.

> How is it handled on cob4?

The topic is published by `cob_safety_controller`, but it is used only on cob4-2 with the old drives.

> cob_omni_drive_controller?

The new base chain does not need this topic anymore, since `ros_canopen reset the controller internally.

ipa-mig, 2016-01-22:
> The new base chain does not need this topic anymore, since ros_canopen resets the controller internally.

> The software controller should reset its internals in order to not jump after recover.

How is this handled therein, i.e. how does the controller know when to reset?

ipa-mdl, 2016-01-22:
> How is this handled therein, i.e. how does the controller know when to reset?

The controller does not know this, but ros_canopen/ros_control reset the controller.
Not only in case of EM stops, but as well for mode switches etc.

ipa-mig, 2016-01-22:
Yes, but what triggers the reset? The EM stop as well? Then, this needs to be known by `ros_canopen`/`ros_control`. How so?

Or am I being stupid and missing something? ;-)

ipa-mdl, 2016-01-22:
> Then, this needs to be known by ros_canopen/ros_control. How so?

This is a different question ;)
The Elmo controller reports the failure via a CANopen emergency message.
For old Elmos and Schunk you get a motor undervoltage error and for the new Elmos you get a Safety stop error :)
`cob_canopen_motor` disabled these error messages for some reason.

ipa-mig, 2016-01-22:
Alright, so the new modules don't rely on the `emergency_stop_state` topic anymore. Understood ;-)

ipa-fxm, 2016-02-04:
Trying to catch up on this:
I guess the approach to follow is what @ipa-bnm wrote in https://github.com/ipa320/cob_robots/issues/403#issuecomment-173558816:
 - none of the raw's have __real__ relayboard
 - raw3-1, raw3-3 and raw3-5 have phidgetboard that is able to publish the required `emergency_stop_state` topic. thus, for them relayboard sim should be removed and phidgetboard added
 - raw3-2, raw3-4 and raw3-6 will use relayboard sim 
 - https://github.com/ipa320/cob_robots/pull/395 needs to be merged
 - modification of `cob_undercarriage_control` is not worth the effort as it is going to be deprecated/removed soon anyway...

@ipa-mig, @ipa-bnm @destogl please confirm or correct me if I'm wrong.....

ipa-mig, 2016-02-04:
as far as I see, you are correct.

I'm unsure if the phidget boards are all the same, are wired correctly/the same or if there are any discrepancies between raw3-1, 3-3 and 3-5.

destogl, 2016-02-04:
You are correct.

ipa-mdl, 2016-02-04:
Seems legit.
I suggest to migrate just the robots that have their EM wiring confirmed.

destogl, 2016-02-05:
configuration of phidget board in #400 is correct for raw3-5

PR #407 (2016-06-10): use phidgets rather than simulated relayboard on raw3-1
(supposed to) fix #403 

This is just a copy of the raw3-3 configuration and needs to be adjusted and tested on raw3-1
@ipa-bnm can you help me here?
closed 2016-06-10 by ipa-nhg