ipa320/cob_robots #422: switch raws and cob3s to new canopen base driver

ipa-fmw, 2016-02-12:
currently done and tested for cob4-2 (see #405).
Instructions see https://github.com/ipa320/cob_robots/pull/405#issuecomment-182945753

@destogl will test with raw3-5

ipa-mdl, 2016-02-12:
It does not work for cob-3, because steer and drive are coupled mechanically.
If we want to support them, first https://github.com/ros-industrial/ros_canopen/issues/101 needs to be addressed and a specialised motor node needs to be written.

destogl, 2016-02-12:
@ipa-fmw: will do. But probably not before march because currently I have other student working on the robot using powercube but I will try to port also the powercube to use SocketCAN (it shouldn't be too hard) and than I test every thing.

ipa-mdl, 2016-02-12:
@destogl: a SocketCAN-based powercube driver sound interesting. `socketcan_interface` already encapsulates most aspect. In addition there is a new topic-base interface (http://wiki.ros.org/socketcan_bridge)

destogl, 2016-02-15:
@ipa-mdl Thanks! I prefer library-like interface for this kind of stuff. We should have already one implementation which is independent of socketcan_interface, I am looking to integrate it in the next days.

ipa-fxm, 2016-03-04:
Take care and note of https://github.com/ipa320/cob_control/issues/98!

andreeatulbure, 2017-01-11:
Hello,

I and @destogl tried to test the ros_canopen config #405 on the raw3-5. We followed the config steps described in #405 
System set-up:
    overlay cob_driver with ipa320/cob_driver#258
    source overlay https://github.com/ros-industrial/ros_canopen.git
    source overlay: https://github.com/ros-controls/ros_control.git (maybe optional)
    disable peak driver and activate socketcan peak_usb
    configure socketcan with bitrate 1000000
Config (based on #417) :
    in base_driver.yaml:
        add missing wheel
        update homing_offset (EncOffsetIncr in base/CanCtrl.ini)
        check homing_event (HomingDigIn in base/CanCtrl.ini)
        set UM=2 ("606A": "1" # read velocity from velocity sensor)
    in base_controller.yaml:
        add missing wheel and update according to base/Platform.ini

The wheels move asynchronously, not synchronously as they do using the old driver. They move to the limit-position, but they stay there and do not move back to zero-position. In the old driver file, CanCtrlPltfCOb3.cpp begining with line 1136, there is some code to move the wheels to zero position but it is written that "// this could be handled also by the elmos themselve". That does not happen in our case. 

We do not receive any errors in the console regarding unsuccessful initialization or something similar. What we receive is:

Loaded joint_state_controller
Loaded twist_controller
Loaded odometry_controller
Started ['joint_state_controller'] successfully
Started ['twist_controller'] successfully
Started ['odometry_controller'] successfully
[base/base_controller_spawner-12] process has finished cleanly

Do you have any idea what might be wrong?

Thanks,
Andreea

ipa-fxm, 2017-01-19:
When working in on #588 I wondered why we still use the "old" base driver for raw3-1 and raw3-3....
I thought we already switched those to use canopen?
That would also give @destogl and @andreeatulbure a better template for raw3-5

@ipa-mdl @ipa-bnm @ipa-mig 
Did we only do this on hardware and not play it back to ipa320?

ipa-mdl, 2017-01-19:
> I thought we already switched those to use canopen?

I don't think so.
Only as a test.

ipa-mig, 2017-01-19:
@ipa-fxm this has only been a test.

ipa-fmw, 2017-01-20:
if the test was succesfull, what prevents us from changing it permanently? would reduce a lot of code overhead with the old base_drive_chain...

ipa-mdl, 2017-01-20:
> if the test was succesfull, what prevents us from changing it permanently? 

These were the very first tests..
Back then we discovered problems with the homing switch.
These should be resolved now, because we have `cob_elmo_homing`.

ipa-fxm, 2017-01-20:
Where are those changes and configs? robot-hw only? raw3-1 or raw3-3? do you remember?

ipa-mdl, 2017-01-20:
> do you remember?

Not really..
I think, this is the full raw config for the controller: https://github.com/ipa320/cob_control/blob/indigo_dev/cob_omni_drive_controller/doc/full_overlay_example.yaml

ipa-mdl, 2017-01-20:
@andreeatulbure: It would be great if you could point us to the config you have tested.

andreeatulbure, 2017-01-24:
Here is the config I used: https://github.com/andreeatulbure/test/tree/indigo_dev/cob_hardware_config/raw3-5/config
For base_controller.yaml I did not use the full_overlay_example file because when using it there have been errors while loading the three controllers.

ipa-mdl, 2017-01-24:
@andreeatulbure:

please add https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_hardware_config/cob4-2/config/base_controller.yaml#L12 to your config

Have you set UM=2?
(https://github.com/ipa320/canopen_test_utils)

rosrun canopen_test_utils canopen_elmo_console can0 1 <<< "UM"
(test for 1 to 6)

andreeatulbure, 2017-01-25:
@ipa-mdl 
I added the line to my config and tested the rosrun command which returned UM=2. But it still does the same thing as before.

andreeatulbure, 2017-01-25:
It is working now

ipa-mdl, 2017-01-25:
[Just finished writing before the last message ;)]

@andreeatulbure: I am sorry, I haven't read your error description properly. I guess I was confused by "limit position" and "zero position" and thought that homing does not work at all.

However, setting the required_drive_mode was correct anyway ^^

> The wheels move asynchronously, not synchronously as they do using the old driver.

This is the intended behaviour.

> They move to the limit-position,

So homing works and the new "zero" position was set.

>  but they stay there and do not move back to zero-position.

Moving to the "neutral" position is just for visual inspection by the user, but is not required for moving the base.
With the new design, moving to the neutral position is not handled in the driver anymore.
It is done  in `twist_controller`, which was started successfully (according to you log output).
The neutral position is the command that is used until the the first twist command was received.

You can adapt it with the `steer_neutral_position` option ([example](https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_hardware_config/cob4-3/config/base_controller.yaml#L31)). Values could be taken from [Platform.ini](https://github.com/andreeatulbure/test/blob/indigo_dev/cob_hardware_config/raw3-5/config/base/Platform.ini#L99)

ipa-fxm, 2017-04-21:
I am willing to have a look at this for `raw3-3` but need assistance from @ipa320/navigation-push (for `sudo` access) as well as maybe @ipa-mdl/@ipa-bnm (for can-driver/canopen support)

ipa-jba, 2017-04-27:
After migrating raw3-3 to canopen a few things we (@ipa-mdl) had to do different than described in https://github.com/ipa320/cob_robots/pull/405#issuecomment-182945753

1) We did not need a source overlay of `ros_control`

2) The whole configuration for the driver was different. The CAN Card we use does not work with the `peak_pci` drivers. Instead we installed the driver supplied by peak (http://www.peak-system.com/fileadmin/media/linux/index.htm) version 8.3.1 and build it with `make NET=NETDEV_SUPPORT` (!! The manual differs from the website in this point. The manual is correct !!). 

3) Adjust the bitrate of the interface for the driver: See the chapter 3.3 in the manual for "Configure Software" (add `options pcan bitrate=1M` to `/etc/modprobe.d/pcan.conf`)

4) After the installation of the new pcan driver load the module with `sudo modprobe pcan` and continue with 
  > configure socketcan with bitrate 1000000: http://wiki.ros.org/socketcan_interface#Initialize_NIC

Note: Right now the `raw3-3` has some issues running with the new drivers

ipa-fxm, 2017-04-27:
> Note: Right now the `raw3-3` has some issues running with the new drivers

Issue is https://github.com/ipa320/cob_robots/pull/661#issuecomment-297448019
I'm on it

ipa-mdl, 2017-04-27:
@ipa-jba: Step 3 should be optional

ipa-fxm, 2017-05-08:
`raw3-3` is migrated....only missing platform is `raw3-1`
@ipa-mig @ipa-jba and @ipa-rmb will sync to find a time slot where the migration could be scheduled...

ipa-rmb, 2017-05-09:
Yes we talked already, we can start on 18.5. but preferably on 22.5.

ipa-fxm, 2017-07-03:
@ipa-rmb @ipa-jba @ipa-mig 
have you scheduled a hackathon for the raw3-1 canopen migration yet?

ipa-mig, 2017-07-04:
not yet. But I guess @ipa-rmb and @ipa-jba need to take the lead on this...

ipa-rmb, 2017-07-11:
As far as I can remember @ipa-jba asked me for possible time slots doing this upgrade. I am aware of the following next presentations of the robot: 19.7., 25.7., and maybe 2.8.
The work can be done at anytime in between with sufficient time until the next demo (and everything should work again afterwards). However, please also ask @ipa-bfb and Falk as well whether they have scheduled work on the robot.

ipa-rmb, 2017-07-11:
Raw3-1 will also be used in the morning of 21.7.

ipa-rmb, 2017-07-11:
I am fine with 26.7.+27.7.

ipa-fxm, 2017-07-13:
@ipa-jba 
are you available on those dates?
(at least ask @ipa-mig about the Elmo tuning procedure!)

ipa-jba, 2017-07-18:
I think I can make it work

ipa-fxm, 2017-07-18:
@ipa-jba FYI
https://github.com/ipa320/care-o-bot/issues/42#issuecomment-316152815

ipa-fxm, 2017-08-03:
raw3-1 is migrated in #702 
which is the last robot (which is still supported) that uses the old "base-drive-chain"-setup

as a follow-up now several launch files and packages become obsolete....cleanup is tracked in new issue

ipa-rmb, 2017-08-04:
We possibly have a filming appointment with Rainer next Wednesday and afterwards the setup of raw3-1 can be updated completely (including the new torso module).

destogl, 2017-08-07:
Hi all,

We still have some small issues that robots gets to edge of stability. Reason for that are controller parameters. Can you give me again link where parameterization process is described? I believe that we are currently the only one using short version of platform, therefore we need a bit different parameters.

Thanks!

ipa-fxm, 2017-08-07:
raw3-3 uses base_short, too
and it's been running for quite a while with canopen now...

destogl, 2017-08-07:
> did you re-run the native Elmo tuning routine in the Elmo software?

No. Can you please point me how to do this?

closed 2017-08-11 by ipa-fxm

ipa-jba, 2017-08-11:
> that robots gets to edge of stability.
Does does the emergency stop / stuck detector trigger, or is it more of a shaking which gets really strong if the robot rotated on place?

We had the shaking but it was solved setting UM=2 for the drive wheels.
Stuck detector is relaxed now and should not trigger that easy (if you are running low on battery for example)

PR #702 (2017-08-11): Migrate raw3-1 base to canopen
fixes #422 (as it is the last robot to be migrated) :tada: 

See discussion in https://github.com/ipa320/cob_robots/issues/422

Disables stuck detector for raw3-1