ipa320/cob_robots #437: [cob_hardware_config] further simplify configuration structure

ipa-fxm, 2016-02-19:
The structure already got much simpler but there still is some potential.

I'm creating this issue for discussion following
 - https://github.com/ipa320/cob_common/pull/175#issuecomment-154128294
 - https://github.com/ipa320/cob_robots/pull/359

This has minor priority for now...

ipa-fxm, 2017-01-19:
After cleaning up unsupported robots #588, we will start introducing a more generic robot-/module-/component-base structure for all URDF, launch and config files... as discussed with @ipa-fmw @ipa-nhg @ipa-mdl 

This will allow further de-duplication of code, more consistency between robots, better harmonization of HW/Sim-behavior as well as easier future maintenance and up-scaling...

ipa-fxm, 2017-03-17:
#622 is a significant step towards a simplified configuration structure

I will apply a similar concept for the other "unifiable" components, e.g. torso, head, sensorring, arm(s),...

ipa-fxm, 2017-03-17:
Also, we would need to adjust the structure to the robot names we are using now to refer to a (physically existing) robot composition, i.e. `cob4-paul-ingolstadt`...following @ipa-tif's spreadsheet Verbauliste.xlsx

The ultimate goal could be to even generate launch files based on a module configuration based on @ipa-mdl's [roslaunch-dump](https://gist.github.com/ipa-mdl/66b14a895e13003a2f76e483bd00aa63)

ipa-fxm, 2017-03-20:
@ipa-mdl asked for input about `roslaunch-dump` (with the final goal being added to `roslaunch`)
https://github.com/ros/ros_comm/issues/1022

ipa-fmw, 2017-04-06:
reminder for @ipa-fxm: extract sick scanners from base urdf

ipa-fxm, 2017-04-12:
> extract sick scanners from base urdf + then remove nav configs for mlr bases as they do not have laser scanners anymore

done in https://github.com/ipa320/cob_robots/pull/654

ipa-fxm, 2017-05-06:
atm none of the solutions proposed by @ipa-fxm and @ipa-nhg fully meet our requirements
also current efforts require a final decision regarding future maintaining responsibilities and supported hardware

thus, I'm moving it to Backlog

ipa-fmw, 2017-07-04:
we won't work on this for now

ipa-mdl, 2017-07-04:
> we won't work on this for now

No need to close it.

ipa-fxm, 2017-07-04:
it's been in the sprint backlog anyway...