ipa320/cob_robots #461: [cob_bringup] env.sh as env-loader script

ipa-fxm, 2016-04-07:
@ipa-nhg @ipa-fmw

This is evolving from https://github.com/ipa320/cob_robots/pull/459#issuecomment-205390813

ipa-fxm, 2016-04-07:
The problem is that we cannot use `$(find cob_bringup)` within the `env-loader` part of the `machine`-tag as it gets resolved __before__ being passed to the remote PC!
Thus, if an overlay of `cob_robots` exists on the b1-PC, "/u/user-name/git/care-o-bot/src/cob_robots/cob_bringup/env.sh" is passed over to the remote PC, e.g. h32. 
However, if no overlay exists on the remote-PC, this file does not exist, and `cob_bringup/robot.launch` will fail...

ipa-fxm, 2016-04-07:
I don't have a good solution yet, but maybe we could put the `env.sh` from `cob_bringup` into a folder that can be accessed by all users, e.g. `/opt/care-o-bot`. We would need to put this file there when setting up a PC afresh...or let the robot users install the `env.sh` into this folder instead of into [`${CATKIN_PACKAGE_SHARE_DESTINATION}`](https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_bringup/CMakeLists.txt#L48-L50)

ipa-nhg, 2016-04-07:
If I understood right , please @ipa-fmw correct me if I am wrong, the use of a 32 bits machine is temporal. The issue with the sound driver and 64 linux system should be solved soon. I don't think that make sense put  effort on that. what about use for the moment the `/opt/ros/indigo/env.sh` ?? or the release version of cob_sound is out-of-date?

My upstart installation also create a env.sh file, that with a full cob installation will be copied to `/etc/ros/$ROS_DISTRO/cob.d/setup/cob.d/setup/env.sh` for all the cob-pcs : https://github.com/ipa320/setup_cob4/blob/master/upstart/upstart_install.sh#L30 . But some modifications will be needed to use it. And from my point of view, is not nice use a global path (hardcoded) on cob_bringup....robot.xml

ipa-fxm, 2016-12-15:
Has this been solved?
Still using the 32bit machine? What about user management?
@ipa-nhg please comment and close (if applicable)

ipa-fmw, 2017-01-10:
in cob4-7 we're not using a 32-bit ubuntu on the head pc any more.

ipa-fxm, 2017-01-19:
I guess then this can be closed...

closed 2017-01-19 by ipa-fxm

ipa-nhg, 2017-01-19:
It can be closed. Actually it is already done, it was needed in case that you source a different workspace and start something remotely, for further info see https://github.com/ipa320/cob_robots/pull/537