ipa320/cob_robots #557: cob4-7: head axis error

ipa-nhg, 2016-11-21:
After init the axis the service returns a success and if I try to init again the component answer that the component is already initialized, I can read the joint_states but I can't move it (the controllers are not properly loaded). Canopen says by init:

[ INFO] [1479722779.349713591]: Initializing XXX
[ INFO] [1479722779.357355224]: Current state: 1 device error: system:0 internal_error: 0 (OK)
[ INFO] [1479722779.357563119]: Current state: 2 device error: system:0 internal_error: 0 (OK)
[ERROR] [1479722780.938229190]: Node has emergency error
illegal tranistion 2 -> 4
Could not switch to mode 7, reason: Could not set transition
[ERROR] [1479722781.568261371]: head_2_jointcould not enter mode 7
illegal tranistion 2 -> 4
Could not switch to mode 7, reason: Could not set transition
[ERROR] [1479722781.568315266]: head_3_jointcould not enter mode 7

And recover:

[ INFO] [1479722806.798563675]: Recovering XXX
illegal tranistion 8 -> 6
[ERROR] [1479722810.948258587]: Node has emergency error

@ipa-mdl do you have any idea where could be the problem?

ipa-mdl, 2016-11-21:
> Canopen RELEASE version

Do not use the release version, it will not work.

> do you have any idea where could be the problem?
 
`Node has emergency error`.
Diagnostics should list the error code(s).
@ipa-tif might help to decode them.

ipa-nhg, 2016-11-21:
That is the problem , diagnostics is green, says that everything is working (?)

ipa-mdl, 2016-11-21:
`candump canX | rosrun canopen_test_utils readable.py elmo_mapping | grep EMCY`

ipa-nhg, 2016-11-22:
INIT:

['can1', 'EMCY', 72, 'EEC:', 'FF10', 'Reg:', '81', 'Msef: ', 'A800000000']
['can1', 'EMCY', 72, 'EEC:', 'FF10', 'Reg:', '81', 'Msef: ', 'A800000000']

@ipa-mdl Thanks

ipa-mdl, 2016-11-22:
FF10=Failed to start motor
0xA8=168=Speed too large to start motor
[according to the manual ;)]

ipa-nhg, 2016-11-24:
@ipa-tif could you please take a look at this issue? Thanks!

ipa-fxm, 2016-12-15:
I guess #568 #566 #561 are related
@ipa-mdl @ipa-nhg please comment and close

ipa-fmw, 2017-03-20:
not relevant any more. will open new tickets concerning new HW and SW versions if neccessary

closed 2017-03-20 by ipa-fmw