ipa320/cob_robots #577: Use CMake-Macro to retrieve sub-dir-list

ipa-fxm, 2017-01-11:
use for ```robotlist``` and ```envlist``` e.g. for install tags in various packages

See http://stackoverflow.com/questions/7787823/cmake-how-to-get-the-name-of-all-subdirectories-of-a-directory

as suggested by @ipa-mdl in https://github.com/ipa320/cob_robots/pull/573#discussion_r93017007

ipa-mdl, 2017-01-11:
File globbing might stop CMake from rebuilding as it does not detect changes..
I think we can export the (hand-crafted) robotlist from ` cob_hardware_config`.
In order to use it, we have to `build_depend` on `cob_hardware_config`.

In `cob_moveit_config` we should install files from the package directory with exclusion rules.

ipa-fxm, 2017-01-19:
exported lists are used in all related packages....

closed 2017-01-19 by ipa-fxm