ipa320/cob_robots #641: base_controller not running correctly in simulation

ipa-fxm, 2017-03-22:
as reported by @ipa-bfb
they seem to be drifting...needs investigation....

ipa-fmw, 2017-03-22:
are you sure this is the base_controller? What I observed was just the simulated robot model slightly drifting in gazebo (independent of wheel positions and controller)

ipa-bfb, 2017-03-22:
try to use cob_teleop and see if it works...

ipa-fmw, 2017-03-22:
sorry, ignore my comment. Seems as the recently introduced commits broke driving the robot in gazebo. we'll have to investigate.

ipa-fxm, 2017-03-22:
Why are the `steer_target_velocity` that high? Is this rad/sec? @ipa-mdl 
```
rostopic echo /base/twist_controller/wheel_commands 
header: 
  seq: 52189
  stamp: 
    secs: 52
    nsecs: 923000000
  frame_id: ''
drive_target_velocity: [0.0, 0.0, 0.0]
steer_target_velocity: [7.60710124957176, -10.0, -10.0]
steer_target_position: [0.0, -3.141592653589793, 0.0]
steer_target_error: [1.395496564844115, -2.722543136645081, -3.048259837630063]
```

ipa-mdl, 2017-03-22:
> they seem to be drifting...needs investigation....

I had the same problem when we implemented the dynamic-reconfigure feature.
For some reason it just works with you setup.

> Is this rad/sec?

Of course ;)

> or we try to spawn the wheels in gazebo at the correct starting position 

+1
The starting postions point away from the base link.
That's why the back wheel has a neutral position of 180 degrees

ipa-fxm, 2017-03-23:
There seems to be something else:
I can set the steer_joints to the desired neutral position (cob4-2) using
```
rosservice call /gazebo/set_model_configuration '{ model_name: "robot", urdf_param_name: "robot_description", joint_names: [ "fl_caster_rotation_joint", "b_caster_rotation_joint", "fr_caster_rotation_joint" ], joint_positions: [ 0.0, -3.141592653589793, 0.0 ] }'
```
the joint_states are then at the desired position, but still the wheels start drifting....I also cannot use `roslaunch cob_teleop teleop_keyboard.launch`

Even if I set remove the `steer_neutral_position` from the yaml, the drifting occurs and I cannot drive...

 > For some reason it just works with you setup.
@ipa-mdl, I don't know why it seemed to work on my setup....

ipa-fmw, 2017-03-24:
this commit seems to have broken moving the base in simulation: https://github.com/ipa320/cob_common/commit/6f6ccc2552e42d57c851997e4bf825fa5ea1c3d0 
If I revert this, simulation is OK again. @ipa-mdl: any idea what might cause the troubles?

ipa-fmw, 2017-03-24:
setting the effort limit back to `100` fixes the issue. see https://github.com/ipa320/cob_common/pull/215

ipa-fmw, 2017-03-24:
@ipa-bfb: FYI fixed.

closed 2017-03-24 by ipa-fmw

ipa-bfb, 2017-03-24:
Awesome! Thanks ;)