ipa320/cob_robots #644: upload_robot.launch

ipa-bfb, 2017-04-01:
I'm having problems with the --inorder tag. Is this suppose to work?

ipa-fxm, 2017-04-01:
what URDF are you trying to upload?

ipa-bfb, 2017-04-01:
Just launching cob4-7

ipa-fxm, 2017-04-01:
what's the error?
we used it on the robot yesterday...

ipa-bfb, 2017-04-01:
Sorry it is working for cob4-5 and cob4-7. The problem is with rob@work:

while processing /home/bfb/catkin_ws/src/cob_simulation/cob_gazebo/launch/robot.launch:
while processing /home/bfb/catkin_ws/src/cob_robots/cob_bringup/robots/raw3-1.xml:
while processing /home/bfb/catkin_ws/src/cob_robots/cob_hardware_config/common/upload_robot.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/indigo/lib/xacro/xacro --inorder '/home/bfb/catkin_ws/src/cob_robots/cob_hardware_config/raw3-1/urdf/raw3-1.urdf.xacro'] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro --inorder '$(arg pkg_hardware_config)/$(arg robot)/urdf/$(arg robot).urdf.xacro'" name="robot_description"/>
The traceback for the exception was written to the log file

ipa-fxm, 2017-04-01:
works for me:
```
roslaunch cob_hardware_config upload_robot.launch robot:=raw3-1
... logging to /home/fxm/.ros/log/07d0a020-16f9-11e7-9d15-00271039ca78/roslaunch-sony-vaio-8513.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global property: pi
when processing file: /home/fxm/projects/indigo/care-o-bot/src/universal_robot/ur_description/urdf/ur10.urdf.xacro
included from: /home/fxm/projects/indigo/care-o-bot/src/cob_common/cob_description/urdf/ur_arm/ur10_arm.urdf.xacro
included from: /home/fxm/projects/indigo/care-o-bot/src/cob_robots/cob_hardware_config/raw3-1/urdf/raw3-1.urdf.xacro
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://www.ros.org/wiki/xacro
 new: http://ros.org/wiki/xacro (/home/fxm/projects/indigo/care-o-bot/src/universal_robot/ur_description/urdf/ur10.urdf.xacro)
started roslaunch server http://sony-vaio:38779/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES

auto-starting new master
process[master]: started with pid [8528]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 07d0a020-16f9-11e7-9d15-00271039ca78
process[rosout-1]: started with pid [8541]
started core service [/rosout]
```
(except for urdf or universal_robot - but this is just a warning)

Check your workspace and update...

ipa-fxm, 2017-04-01:
you can get more information by using 
`roscd cob_hardware_config/raw3-1/urdf &&  rosrun xacro xacro --inorder raw3-1.urdf.xacro`

ipa-bfb, 2017-04-01:
Solved! Thanks for the help ;) Have a nice weekend and if you are in for a run let me know ;)

closed 2017-04-01 by ipa-fxm