ipa320/cob_robots #656: cob3-2 tray not moving

ipa-nhg, 2017-04-18:
init: ok
joint_states: OK

parameter config:
```
rosparam get /tray/tray_driver/
bus: {device: can0}
defaults: {eds_file: cob3-9/config/Schunk_0_63.dcf, eds_pkg: cob_hardware_config}
modules:
- {id: 10, name: tray_joint}
name: tray_controller
sync: {interval_ms: 10, overflow: 0}
```

bringup log:
```
roslaunch cob_bringup canopen_402.launch robot:=cob3-9 can_module:=can0 component_name:=tray
... logging to /u/test/.ros/log/06d8df8c-243e-11e7-851f-00c088010820/roslaunch-cob3-2-pc3-8329.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cob3-2-pc3:48489/

SUMMARY
========

CLEAR PARAMETERS
 * /tray/tray_driver/

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.10
 * /tray/joint_group_position_controller/joints: ['tray_joint']
 * /tray/joint_group_position_controller/required_drive_mode: 1
 * /tray/joint_group_position_controller/type: position_controll...
 * /tray/joint_group_velocity_controller/joints: ['tray_joint']
 * /tray/joint_group_velocity_controller/required_drive_mode: 2
 * /tray/joint_group_velocity_controller/type: velocity_controll...
 * /tray/joint_names: ['tray_joint']
 * /tray/joint_state_controller/publish_rate: 50
 * /tray/joint_state_controller/type: joint_state_contr...
 * /tray/joint_trajectory_controller/action_monitor_rate: 10
 * /tray/joint_trajectory_controller/constraints/goal_time: 0.6
 * /tray/joint_trajectory_controller/constraints/stopped_velocity_tolerance: 0.05
 * /tray/joint_trajectory_controller/constraints/tray_joint/goal: 0.1
 * /tray/joint_trajectory_controller/constraints/tray_joint/trajectory: 0.1
 * /tray/joint_trajectory_controller/joints: ['tray_joint']
 * /tray/joint_trajectory_controller/required_drive_mode: 7
 * /tray/joint_trajectory_controller/state_publish_rate: 25
 * /tray/joint_trajectory_controller/stop_trajectory_duration: 0.5
 * /tray/joint_trajectory_controller/type: position_controll...
 * /tray/max_command_silence: 0.5
 * /tray/tray_driver/bus/device: can0
 * /tray/tray_driver/defaults/eds_file: cob3-9/config/Sch...
 * /tray/tray_driver/defaults/eds_pkg: cob_hardware_config
 * /tray/tray_driver/modules: [{'name': 'tray_j...
 * /tray/tray_driver/name: tray_controller
 * /tray/tray_driver/sync/interval_ms: 10
 * /tray/tray_driver/sync/overflow: 0
 * /tray/tray_joint_position_controller/joint: tray_joint
 * /tray/tray_joint_position_controller/required_drive_mode: 1
 * /tray/tray_joint_position_controller/type: position_controll...
 * /tray/tray_joint_velocity_controller/joint: tray_joint
 * /tray/tray_joint_velocity_controller/required_drive_mode: 2
 * /tray/tray_joint_velocity_controller/type: velocity_controll...

NODES
  /tray/
    control_mode_adapter (cob_control_mode_adapter/cob_control_mode_adapter_node)
    tray_driver (canopen_motor_node/canopen_motor_node)
    tray_joint_state_controller_spawner (controller_manager/spawner)

ROS_MASTER_URI=http://cob3-2-pc1:11311

core service [/rosout] found
process[tray/tray_driver-1]: started with pid [8338]
process[tray/tray_joint_state_controller_spawner-2]: started with pid [8351]
process[tray/control_mode_adapter-3]: started with pid [8355]
[ INFO] [1492524961.870238687]: waitForService: Service [/tray/controller_manager/load_controller] has not been advertised, waiting...
[ WARN] [1492524961.987274700]: falling back to 'modules', please switch to 'nodes'
[ INFO] [1492524962.231005088]: Using fixed control period: 0.010000000
[ INFO] [1492524966.874551853]: waitForService: Service [/tray/controller_manager/load_controller] has not been advertised, waiting...
[ INFO] [1492524971.885885665]: waitForService: Service [/tray/controller_manager/load_controller] has not been advertised, waiting...
[ INFO] [1492524973.019234962]: Initializing XXX
[ INFO] [1492524973.019536821]: Current state: 1 device error: system:0 internal_error: 0 (OK)
[ INFO] [1492524973.019708655]: Current state: 2 device error: system:0 internal_error: 0 (OK)
[ INFO] [1492524975.228286526]: waitForService: Service [/tray/controller_manager/load_controller] is now available.
[ INFO] [1492524975.595409405]: joint_trajectory_controller loaded
[ INFO] [1492524975.815247719]: joint_group_position_controller loaded
[ INFO] [1492524975.945279465]: joint_group_velocity_controller loaded
[ INFO] [1492524975.955224521]: Switched Controllers. From no_stop_controller_defined to joint_trajectory_controller
```

candump: https://gist.github.com/ipa-nhg/93962016397cad8f76a223f77919467a

ipa-mdl, 2017-04-18:
[cob3-2_tray_not_moving.txt](https://github.com/ipa320/cob_robots/files/930355/cob3-2_tray_not_moving.txt)

> Switched Controllers. From no_stop_controller_defined to joint_trajectory_controller

The hardware did not switch.
`ros_control` or `ros_canopen` is outdated.

ipa-nhg, 2017-04-19:
@ipa-mdl thank you for your help. ros_canopen is updated ( release version ), for ros_control I updated some packages but not the whole metapackage. I will try this , I hope an update of ros_control doesn't affect to the schunk_powercube_chain driver installed on cob3-2...

ipa-mdl, 2017-04-19:
>  ros_canopen is updated ( release version ),

Release version is fine, no overlay is needed.
Please  check that `canopen_motor_node` got updated.
(updating the metapackage does not update the individual packages)

>  for ros_control I updated some packages but not the whole metapackage.

I guess you will need  `controller_manager` and `hardware_interface` at least.

> I hope an update of ros_control doesn't affect to the schunk_powercube_chain

It is completely unrelated :)

ipa-nhg, 2017-04-19:
@ipa-mdl thanks a lot! updated ``controller_manager`` and ``hardware_interface``solved the problem with the tray,  it was also needed a new MTS configuration,  this PRL+ build had a different  configuration (not inverted motor,  new limits , new reference position...).

The tray is running and successfully  tested with the default config positions, buttttttt now the  torso is not working, I am not sure if the update of ros_control produced the error, or the torso didn't work before, I tried to debug the problem and at least the namespaces of the trajectory controller topics are wrong, maybe our current NS version have never been tested with the old Schunk modules... @ipa-fmw do you have time to check the problem with me? (four eyes see more than two)

ipa-fmw, 2017-04-20:
we agreed to stop here and not further debug cob3-2. @ipa-uhr will communicate that to khenglee

closed 2017-04-24 by ipa-fmw