ipa320/cob_robots #669: review camera coordinate frames

ipa-fmw, 2017-05-15:
there's a mismatch between `_link` and `_optical_frame` for the cameras we're using in cob_bringup
- realsense: OK, compliant with REP103 http://www.ros.org/reps/rep-0103.html (done in https://github.com/ipa320/cob_robots/pull/614)
- Asus/kinect: Not OK
- Sick Visionary: Not Ok
- usb head cam: to be verified

ipa-fxm, 2017-09-15:
- intel realsense r200 and zr300 are conform to the standard but might have wrong mount positions in our URDFs...this has been fixed for cob4-12 (in https://github.com/unity-robotics/unity_robots/pull/8) and cob4-7 (in https://github.com/ipa320/cob_robots/pull/720)

 - asus camera did not follow standard....this was fixed in https://github.com/ipa320/cob_robots/pull/720 and tested on cob4-7

**ToDo:**
 - [ ] verify Sick Visionary/3dcs
 - [ ] verify USB cam

**ToDo:**
apply fixes also to the other robots affected:
 - [ ] cob4-2
 - [ ] cob4-5
 - [ ] cob4-8
 - [ ] cob4-10

@ipa-nhg @cagbal @ipa-fmw @ipa-rmb FYI

ipa-fxm, 2017-09-18:
all done in #720 and ipa320/cob_common#233

@ipa-rmb @ipa-fmw @ipa-jba @ipa-fez please review/re-test

ipa-fmw, 2017-09-20:
tested on cob4-12: OK

ipa-fxm, 2017-09-21:
PR's https://github.com/ipa320/cob_robots/pull/720 and https://github.com/ipa320/cob_common/pull/233 are good to merge

I've created follow-up Issues to track remaining issues:
https://github.com/ipa320/cob_common/issues/234
https://github.com/SICKAG/sick_visionary_t/issues/19

ipa-fxm, 2017-09-21:
@ipa-fmw 
https://github.com/unity-robotics/unity_robots/pull/10 finalizes this PR series for `cob4-12`

PR #720 (2017-09-26): fix camera coord frames for all cameras and all robots for hw and sim
fixes #669
fixes ipa320/cob4#553
requires https://github.com/ipa320/cob_common/pull/233

includes #715

@ipa-nhg @cagbal @ipa-fmw @ipa-rmb FYI
closed 2017-09-26 by ipa-fxm