ipa320/cob_robots #682: Error while doint moveit setup assistance

ipa-mjp, 2017-07-15:
Hello,

I am getting below error when I was doing moveit setup assistance for care-o-bot4. For moveit setup assistance, I am loading cob4-5.urdf.xacro. Any one can help me to get rid of this error.

[image]

Thank you.

ipa-fxm, 2017-07-16:
duplicate of https://github.com/ipa320/cob_manipulation/issues/105...

closed 2017-07-16 by ipa-fxm

ipa-fxm, 2017-07-17:
for the time being simply run:
`rosrun xacro xacro --inorder raw3-1.urdf.xacro > raw3-1.urdf`
within `cob_hardware_config/robots/urdf` then load the urdf  - and not the xacro from the setup assistant...

ipa-bfb, 2017-07-18:
We want to create the moveit config for cob4 not raw3-1.

Is it the same procedure?