ipa320/cob_robots #781: cob_bringup 0.6.11 in kinetic fails on debian builds

tfoote, 2019-08-06:
From: http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__cob_bringup__ubuntu_xenial_amd64__binary/
```
14:25:33 CMake Error at /opt/ros/kinetic/share/roslaunch/cmake/roslaunch-extras.cmake:30 (message):
14:25:33   roslaunch_add_file_check() path
14:25:33   '/tmp/binarydeb/ros-kinetic-cob-bringup-0.6.11/robots/cob4-22.xml' was not
14:25:33   found
14:25:33 Call Stack (most recent call first):
14:25:33   CMakeLists.txt:14 (roslaunch_add_file_check)
14:25:33 
```

My guess is that the launch file is not installed

tfoote, 2019-08-06:
Digging in a little bit it looks like this package is one of a few cob packages that wasn't updated. Maybe it's cross referenced but needs an update itself following the recent releases. 

https://github.com/ros/rosdistro/pull/21922
https://github.com/ros/rosdistro/pull/21924
https://github.com/ros/rosdistro/pull/21926
https://github.com/ros/rosdistro/pull/21927.
https://github.com/ros/rosdistro/pull/21929
https://github.com/ros/rosdistro/pull/21931
https://github.com/ros/rosdistro/pull/21933
https://github.com/ros/rosdistro/pull/21935
https://github.com/ros/rosdistro/pull/21937
https://github.com/ros/rosdistro/pull/21940

fmessmer, 2019-08-07:
yes, `cob_robot` will get a new release, too, fixing this issue
I wasn't able to do it all in one day...sorry about that...I'll be back at it again in a few hours...

fmessmer, 2019-08-07:
is back to normal

closed 2019-08-07 by fmessmer